openhumanoids/pronto #4: cmake file needs to explicitly locate boost libs

RussTedrake, 2015-01-13:
fyi - when compiling on the mac, I see

```
-------------------------------------------
-- visualization
-------------------------------------------
[  0%] Built target lcmgen_c
[  0%] Built target lcmgen_cpp
[  0%] Built target lcmgen_java
[  0%] Built target lcmgen_python
[ 71%] Built target lcmtypes_visualization
[ 80%] Built target lcmtypes_visualization_jar
[ 85%] Built target vs_vis
[ 90%] Built target collections_renderer
Linking CXX executable ../../bin/collections_example
ld: library not found for -lboost_thread
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

mauricefallon, 2015-01-13:
I see this:
-- checking for module 'lcm'
--   found lcm, version 1.0.0
-- Found Java: /usr/bin/java (found version "1.7.0.65") 
-- Found PythonInterp: /usr/bin/python (found version "2.7.6") 
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   thread
--   system

CMakeLists.txt finds boost like this:
find_package(Boost COMPONENTS thread system REQUIRED)
message(STATUS ${Boost_LIBRARIES})
...
target_link_libraries(vs_vis ${Boost_LIBRARIES})

- At the start of a clean build of visualization what does cmake say?
- It common enough to explicitly list it in target_link_libraries() does that work for mac?

The bug report is appreciated!

RussTedrake, 2015-01-13:
fwiw - i get pages of cmake errors of the form
```
CMake Warning (dev) at CMakeLists.txt:23 (add_library):
  Policy CMP0038 is not set: Targets may not link directly to themselves.
  Run "cmake --help-policy CMP0038" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  Target "vs_vis" links to itself.
```
after filtering through those, i do see
```
BUILD_PREFIX: /Users/russt/drc/software/build

-- Boost version: 1.55.0
-- Found the following Boost libraries:
--   thread
--   system
-- /usr/local/lib/libboost_thread-mt.dylib/usr/local/lib/libboost_system-mt.dylib
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/russt/drc/software/pronto/visualization/pod-build

```

patmarion, 2015-01-13:
The issue is in pronto/visualization/src/collections_viewer/CMakeLists.txt.  It uses the command:

link_libraries(boost_thread)

instead of target_link_libraries(collections_viewer ${Boost_LIBRARIES})

patmarion, 2015-01-13:
These issues only appear on Mac because boost and most other libraries are installed to /usr/local via homebrew.  In older versions of Xcode, /usr/local was in the default search path for the compilers, but now it's not.  That's why you should always prefer to use find_library() and target_link_libraries() in cmake.

mauricefallon, 2015-01-13:
Pat's suggested change works for me (linux), what about you Russ?

@@ -5,11 +5,10 @@ if(NOT BOT2_VIS_FOUND)
     return()
 endif(NOT BOT2_VIS_FOUND)
 
 #add_definitions(-Wall -std=gnu99)
 
-link_libraries(boost_thread)
 
 add_executable(collections_example collections_example.cpp)
 pods_use_pkg_config_packages(collections_example lcm)
 
 # make executables public
@@ -17,6 +16,8 @@ pods_install_executables(collections_example)
 
 add_executable(collections_viewer collections_viewer.cpp)
 pods_use_pkg_config_packages(collections_viewer bot2-vis 
   bot2-lcmgl-renderer lcm collections_renderer)
 
+target_link_libraries(collections_viewer ${Boost_LIBRARIES})
+
 pods_install_executables(collections_viewer)

RussTedrake, 2015-01-14:
did you commit it?  i don't see it in the logs for master?

mauricefallon, 2015-01-14:
I was waiting for you to verify on your end.

Anyway, I just pushed a commit for this to pronto and a pull request to pull it into drc
Can you verify on your end and accept the drc pull request?

closed 2015-01-14 by patmarion