openhumanoids/pronto #48: Fixed bias for imu state

mcamurri, 2016-03-16:
I would like to introduce a ins modality in which I can pre-compute  the gyro and accel bias of the IMU and keep them as fixed and never changed state. 

I know that this is feature interesting also for other users in this organization, but before working on that I would like to know from which branch should I work on, because pronto-distro points to a very old commit of the master branch of this repository, and if I try to do a submodule update against the latest version, I get compilation errors, probably due to name and lcmtypes changes.

Please let me know.

mauricefallon, 2016-03-16:
thanks for the prompt @mcamurri . @raluca-scona  is working close this issue currently and we want to this feature for Valkyrie.

I will fix the pronto-distro build and get in touch soon.

I also need to add a travis-ci build server for pronto-distro. I'll look into this.

mcamurri, 2016-03-16:
I'm implementing a preliminar version right now.
In particular, I'm implementing two modalities:

1. Online initialization of gyro and accel bias
2. Static initialization of gyro and accel bias (from config file)

I studied the code, and I found no specific corrections for random walk errors and other types of error that commonly affect an IMU gyro (some examples are described [here](http://www.vectornav.com/support/library/gyroscope). 

Is @raluca-scona working on this?

mauricefallon, 2016-03-16:
That seems fine. @raluca-scona isn't implementing this as yet, so you are not conflicting. 

Random walk can be estimated. By making conflicting rotation or rotation rate measurements the bias can be learned as it will conflict with the rotation rates from the IMU.

We used this mode with Atlas: when the robot was standing still we could make rotation measurements to learn the bias and avoid rotation drift.

mcamurri, 2016-03-17:
which is the thing you do [here](https://github.com/openhumanoids/pronto/blob/bbcce4a16692592f45e78d32065cb6f5e8d8a0fa/mav/mav-state-estimator/src/mav_state_est/sensor_handlers.cpp#L263), I guess. I was wondering if it was worth trying more sophisticated models, as explained [here](https://www.google.it/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=0ahUKEwj-i-3qxMfLAhULCnMKHU6EAmkQFggcMAA&url=http%3A%2F%2Fieeecss.org%2FCSM%2Flibrary%2F2010%2Ffeb10%2F06-AsktheExperts.pdf&usg=AFQjCNE6NbX1vMDh75p7XKQXBmLat5VStQ&bvm=bv.117218890,d.bGQ) for instance.

Anyway. I've made a [pull request for the issue](https://github.com/openhumanoids/pronto/pull/49).

mcamurri, 2016-03-22:
Modifications has been merged. I close the issue.

closed 2016-03-22 by mcamurri