openhumanoids/pronto #67: wrong acceleration estimation from IMU?

mcamurri, 2016-05-25:
I've noticed here:

https://github.com/openhumanoids/pronto/blob/753361b9bf56b04fd07f9ad5f98416567d67a6b8/state-estimator/src/mav_state_est/sensor_handlers.cpp#L100

that the base acceleration is computed as an the body-imu transform applied to the acceleration vector provided by the IMU.

If this is true, then the code is wrong, because the correct way to compute the acceleration is: rotated imu acceleration + translational part of the transform times the rotational acceleration.

Is this done somewhere else in the code? Am I missing something?

mauricefallon, 2016-07-06:
I agree that the use of bot_trans_apply_vec is very likely to be wrong. 

The end effect of this here would be to induce an incorrect pitch or roll in the output of the estimator.
In fact it may be the course of such an issue I've seen in our own experiments with Valkyrie.

Taking your equation:
* rotated imu acceleration + translation of the transform x rotational acceleration - rotation rate x (rotation rate x translation)
* a + b + c

How would you like to fix (and test) this?
- for a: remove the addition of the offset by only using bot_quat_rotate_to()
- for b: given we don't have a measurement for rotational acceleration, we could treat this term to be zero?
- for c: the gyro measurement can easily be rotated into the body frame and this translation applied.

For b and c, it should be possible to do an analysis of the typical magnitude of these terms.
For example how do they compare to the -9.81 gravity vector. They may be neglibible.

**NOTE: The original authors (Byr and Bacharach) typically used the IMU sensor frame
as the origin for state estimation.** Then this error wouldn't have been present.
This also has the added advantage of removing the imprecision of the measurement of 
the transform between base and imu. Another advantage is that IMU biases need be
measured in the imu frame (as they are physical offsets in the individual acceleration and gyro sensors), and if the frame used for state estimation is different these biases will not be readily observable.

mauricefallon, 2016-07-06:
For clarity below is the function which is being used and the addition of the body-to-imu translation is incorrect:

void
bot_trans_apply_vec(const BotTrans * btrans, const double src[3],
        double dst[3])
{
    bot_quat_rotate_to(btrans->rot_quat, src, dst);
    dst[0] += btrans->trans_vec[0];
    dst[1] += btrans->trans_vec[1];
    dst[2] += btrans->trans_vec[2];
}

mcamurri, 2016-07-06:
I've already implemented the code for all a, b and c.
For now it seems that just removing the translation part in the apply transform function is the most beneficial thing. 
About the other two terms, I need to investigate more.

mauricefallon, 2016-07-20:
An update on this, I applied this change i.e.

```
- bot_trans_apply_vec(&ins_to_body, msg->accel, body_accel);
+ bot_quat_rotate_to (ins_to_body.rot_quat, msg->accel, body_accel);
```

and then tested with logs containing VICON motion capture for Valkyrie. I see only marginal or insignificant change in the pitch and roll versus VICON. 

As a result I'm not going to integrate the change. I'm happy to make the change if you test it on your end and see an improvement.

mcamurri, 2016-07-29:
OK, but I would recommend your change anyway, at least to have a correct estimation of accelerations, which affects also linear velocity and position in the prediction phase, I guess?

mauricefallon, 2017-02-10:
Fixed in https://github.com/openhumanoids/pronto/commit/4593e2d95716ee2e9567d858aa1e6c5a53dd1110

closed 2017-02-10 by mauricefallon