openhumanoids/pronto #81: Extend VO integration to pronto

mauricefallon, 2016-11-12:
* Extend pronto to support orientation corrections from VO. currently just linear translation. straightforward
* support situations when the VO estimate fails entirely, by resetting the measurement (deltaroot)
* get the module to run on @simalpha 's computer or have dls-fusion run on my computer with the hyq kinematics for a trot. this is a key step
* ask @MarcoCar if dls-fusion extends pronto-distro or over writes it?
  * preferable his changes shouldn't change pronto itself, pronto distro is fine.
  * have him use the most recent version of pronto with the VO fusion mode
  
* In the vo algorithm Remove image buffer copy in featureAnalysis() when not outputting features
* IMPORTANT Fix this issue.
Eigen::Isometry3d delta_body_from_ref =  world_to_body_ * ( deltaroot_body_pose_.inverse() );
* IMPORTANT: mention that the acceleration signals should be filtered during foot impacts E.g. with a high gain FIR filter
* TODO: get running on the handheld hyq logs, or any logs with an imu and multisense e.g. husky

MarcoCar, 2016-11-14:
I think you meant @mcamurri again

mcamurri, 2016-11-14:
dls-fusion is located in a separated repo which [our fork of pronto-distro](https://github.com/iit-DLSLab/pronto-distro). It is redefinining the legodo module, and the binary is linking against this module instead of yours.

We anyway use a  [modified version of pronto](https://github.com/iit-DLSLab/pronto/tree/4259ac5f364ed748ebfc822eb538d6da056b3c01), because there are some things  which are not yet pushed to pronto, but I can merge it with your pronto easily and I will do it in  a moment .

Since pronto (not pronto-distro) is the real important module, in the future I think I will create more rational container for pronto and other modules.

mauricefallon, 2016-11-14:
I think it would help if you could use pronto with as few modifications as possible
- Ideally legodo (for bipeds) should have no modifications
- you leg odometry (for quads) could be renamed legodo_quad

Is that possible?

mauricefallon, 2016-11-14:
You should add legodo_quad as a pod in pronto-distro as well as dls-fusion.
I'll look into compiling dls-fusion on my machine with @simalpha 's help later

mcamurri, 2016-11-14:
To add a legodo_quad we need first to add it here:

https://github.com/openhumanoids/pronto/blob/master/state-estimator/src/mav_state_est/rbis_update_interface.cpp#L7

after that, I should change the name everywhere in dls-state-estimator repository and then it is done already.

The modifications we did to pronto (which haven't been pullrequested yet) are the following:

- removing the neck joint for the create octomap process (otherwise I had to send a constant transform to ensure I can build the octomap) 4259ac5
- changing the scan name for the hokuyo (we have two HOKUYOs) 7680fe
- removing a cmake command which was requiring superuser privileges and was not compiling 5ccae52
- implementing the acceleration fix (see #67 )
- implementing the remove-history feature (still experimental, not really used) 1d164fa

what I usually do is to rebase them on top of the latest master from openhumanoids, to stay on track and slowly pull request them.

mcamurri, 2016-11-14:
BTW, I'm including the Fovis in dls-fusion, so I can also try the VO with the logs and produce some statistics.

mauricefallon, 2016-11-14:
[image]

Integrating VO into HyQ/pronto is going to be a very big improvement. It's working a lot better than I expected. After 280 seconds of **trotting**:
- 90% of the log is essentially drift free. (you can see it from the horizontal lines. green is ground truth, white state estimation)
- nearly all the drift is during one 40 second period of trotting in place (on the right of the image). I'm sure we can fix this.

This worked 'out of the box'. There are lots of options for improvement and refinement.

mcamurri, 2016-11-14:
how did you include the Fovis in the estimate? This requires to modify the source of dls-fusion, but I don't think I've pushed this already.

mauricefallon, 2016-11-14:
i did it myself, the changes were about 10 lines:
https://gitlab.advrcloud.iit.it/dls-lab/dls-state-estimator/compare/no-ethz...mf-adding-vo-to-dls-pronto

mauricefallon, 2016-11-14:
drift for a 350 second crawl log is almost zero - 9 cm or so . crawl lateral 2016-05-23
(orientation from vicon is used, but I think we can achive this without too)

mauricefallon, 2016-11-15:
@mcamurri  I've added a branch with the vo dependencies to your pronto distro:
https://github.com/iit-DLSLab/pronto-distro/compare/dls-experimental...iit-DLSLab:mf-adding-vo-to-dls-pronto?expand=1

To build:
$ cd visual-odometry
$ make -j

Please try to build it, it should work. I'll have Simona try it out too this week.

mauricefallon, 2016-11-15:
You will also find a procman file here with the launch arguments I used with the trot and crawl logs:
pronto-distro/config/hyq/hyq_vo.pmd

Only need to run se-simple-vo in addition to dls-fusion . You currently need to playback logs at x0.5 speed as the VO needs to be optimised.

mcamurri, 2016-11-15:
Thanks @mauricefallon I will try soon and let you know.

mauricefallon, 2016-11-15:
Can put all of the logs for your kinematic estimator into a directory - rather than dumping them at ~/ ?

mcamurri, 2016-11-15:
something like ~/dls-logs/ would be fine? 
Or you mean a folder the user can select from command line arguments?

You can de-activate the logging by setting debug false in the config file BTW

mauricefallon, 2016-11-15:
`something like ~/dls-logs/ would be fine? 
`
perfect.

BTW there a performance benefit to be had by reducing LCM and file i/o when operating pronto. you will not get optimal results if you run do all this i/o online

mauricefallon, 2016-11-16:
@mcamurri 

change you instructions to be:
export PATH=$PRONTO_ROOT/build/bin:$PATH
to put your versions of pronto in front of any others

mauricefallon, 2016-11-16:
i make a variety of changed and now the log playback channels dont need to be changed

mauricefallon, 2016-11-16:
@mcamurri 

can you leave the mems IMU on the robot and collect both it and KVH? it would make for an interesting comparison to see how much better KVH is.

mcamurri, 2016-11-17:
Replying in order:
- Is there a particular reason for putting PRONTO_ROOT at the beginning of the path? I had problems doing that in the past, because custom defined paths in pronto were conflicting with other software like Gazebo. It took a while to figure this out.
- Thank you for the modification, are they picking the channels automatically or should I specify somewhere the channel names (e.g. command line args)?
- I agree. I have to talk to our technician, but I'm afraid it is not possible for two reasons: the mounting positions are conflicting, and the board is using the last ethernet port in the chain. I know they are preparing a board which two ports so you can continue the chain. I can try to record some logs with one IMU and some with the other IMU if it makes sense.

mauricefallon, 2016-11-17:
now at https://gitlab.advrcloud.iit.it/dls-lab/dls-state-estimator/issues/2

closed 2016-11-17 by mauricefallon