openhumanoids/pronto #87: Timestep dt check

mcamurri, 2016-11-24:
It would be nice to have a way to warn the user when the timestep_dt value differs too much from the actual data frequency.

Something like computing the average of the diff of timestamps and compare to the value.

When switching between different IMUs and drivers, when pronto is in the loop it becomes dangerous when the integration step is wrong.

I've already tried in the past to detect the timestep_dt automatically, but it didn't go well.

Maybe when  doing the bias detection we could detect the frequency as well and round to the most reasonable frequency.

mauricefallon, 2016-11-24:
I'm not sure what you mean and where you want to put something. suggest a place and purpose.

mcamurri, 2016-11-25:
Sorry I didn't specify the context. 

I was talking about the IMU handler.

I suggest to collect [here](https://github.com/openhumanoids/pronto/blob/master/state-estimator/src/mav_state_est/sensor_handlers.cpp#L290) or [here](https://github.com/openhumanoids/pronto/blob/master/state-estimator/src/mav_state_est/sensor_handlers.cpp#L98) a certain number of messages (say 100), and compute of the average deta time from the timestamps.

If this number is too far from the user provided parameter `timestep_dt`, the system should warn the user or override the parameter.

Purpose of this suggestion is to avoid badly configured parameters to affect the estimate, especially when used in the loop, in case the IMU frequency changes.

In our specific case, when switching from 250 Hz to 1 kHz IMU without changing the `timestep_dt` parameter, it took a while to figure out why the estimate was wrong and the robot was behaving badly.

I thought it would be a useful feature.

closed 2017-10-09 by mcamurri