openhumanoids/pronto #107: Laser Utils linking issue

gtinchev, 2017-09-26:
Apparently https://github.com/openhumanoids/common_utils/pull/12 breaks the linking of the state estimator.

@mauricefallon I'm struggling to find the reason why the following happens:

```
gtinchev@beatrix:~/rpg-navigation/software/pronto/state-estimator$ make
[ 21%] Built target mav-state-est
[ 24%] Linking CXX shared library ../../lib/libgpf-laser-lib.so
CMakeFiles/gpf-laser-lib.dir/LaserLikelihoodInterface.cpp.o: In function `laser_create_projected_scan_from_planar_lidar_with_interpolation':
LaserLikelihoodInterface.cpp:(.text+0x110): multiple definition of `laser_create_projected_scan_from_planar_lidar_with_interpolation'
CMakeFiles/gpf-laser-lib.dir/laser_gpf_lib.cpp.o:laser_gpf_lib.cpp:(.text+0x12e0): first defined here
CMakeFiles/gpf-laser-lib.dir/LaserLikelihoodInterface.cpp.o: In function `laser_update_projected_scan_with_interpolation':
LaserLikelihoodInterface.cpp:(.text+0x120): multiple definition of `laser_update_projected_scan_with_interpolation'
CMakeFiles/gpf-laser-lib.dir/laser_gpf_lib.cpp.o:laser_gpf_lib.cpp:(.text+0x12f0): first defined here
CMakeFiles/gpf-laser-lib.dir/rbis_gpf_update.cpp.o: In function `laser_create_projected_scan_from_planar_lidar_with_interpolation':
rbis_gpf_update.cpp:(.text+0x6c0): multiple definition of `laser_create_projected_scan_from_planar_lidar_with_interpolation'
CMakeFiles/gpf-laser-lib.dir/laser_gpf_lib.cpp.o:laser_gpf_lib.cpp:(.text+0x12e0): first defined here
CMakeFiles/gpf-laser-lib.dir/rbis_gpf_update.cpp.o: In function `laser_update_projected_scan_with_interpolation':
rbis_gpf_update.cpp:(.text+0x6d0): multiple definition of `laser_update_projected_scan_with_interpolation'
CMakeFiles/gpf-laser-lib.dir/laser_gpf_lib.cpp.o:laser_gpf_lib.cpp:(.text+0x12f0): first defined here
collect2: error: ld returned 1 exit status
src/gpf/CMakeFiles/gpf-laser-lib.dir/build.make:172: recipe for target 'lib/libgpf-laser-lib.so' failed
make[3]: *** [lib/libgpf-laser-lib.so] Error 1
CMakeFiles/Makefile2:288: recipe for target 'src/gpf/CMakeFiles/gpf-laser-lib.dir/all' failed
make[2]: *** [src/gpf/CMakeFiles/gpf-laser-lib.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:27: recipe for target 'all' failed
make: *** [all] Error 2
```

I'm unsure why it is trying to define the function `laser_create_projected_scan_from_planar_lidar_with_interpolation` when it is part of an external library.
I'm unsure if that's an issue of overlapping header files.

Could you please assist?

P.S. I tried changing the `laser_create_projected_scan_from_planar_lidar_with_interpolation` to `laser_create_projected_scan_from_planar_lidar_with_interpolation_with_timeout` and add an argument for the timeout (in the cpp files), however, the same error occurs. Thus, I think it must be something to do with the linking of the header files.

mauricefallon, 2017-09-26:
not sure. could be that your new version of laser_create_projected_scan_from_planar_lidar_with_interpolation() is defined outside of an ifdef/undef  so that the complier sees multiple versions of it when linking against several libraries.

try simply defining it in the hpp and implementing it in the cpp (even if its only one line and see if the error goes away?

you might also be able to define in the hpp as `static inline` ... I'm not sure what does does exactly

gtinchev, 2017-09-27:
You were right, putting the implementation (1-liner) in the .c file did the job. I'll create a separate PR. Thanks!

closed 2017-09-27 by gtinchev